open-rmf/rmf_task #6: Allow all battery considerations to be disabled

mxgrey, 2021-04-07:
## Feature request

### Description

When running simulation tests of traffic management, it can be useful to have the robots do very very long driving tasks non-stop without any need to charge the battery.

Battery draining can be disabled in the simulation, but the task planner will still reject any tasks that it believes will drain the battery too much. For those cases, it would be useful to disable the battery constraint altogether so the planner does not reject those requests.

Yadunund, 2021-04-07:
To clarify, there is a `FleetUpdateHandle::account_for_battery_drain()` [method](https://github.com/open-rmf/rmf_ros2/blob/29244fca1de0cf6f248890b9ccd5caa1f5366d79/rmf_fleet_adapter/include/rmf_fleet_adapter/agv/FleetUpdateHandle.hpp#L105) that can be used to toggle battery considerations when performing task allocation planning. If set `false`, battery drain will not be considered when planning.The `full_control` adapter takes in an [argument](https://github.com/open-rmf/rmf_ros2/blob/29244fca1de0cf6f248890b9ccd5caa1f5366d79/rmf_fleet_adapter/launch/fleet_adapter.launch.xml#L37) to indicate whether battery drain should be considered or not. The implementation is [here](https://github.com/open-rmf/rmf_ros2/blob/29244fca1de0cf6f248890b9ccd5caa1f5366d79/rmf_fleet_adapter/src/full_control/main.cpp#L922-L925)

mxgrey, 2021-04-11:
Thanks for pointing that out. I thought I already tried using that parameter, but I must have messed something up. I tried it again just now, and it's working :+1:

closed 2021-04-11 by mxgrey