open-rmf/rmf_task #42: Support tasks that have indefinite/permanent duration

mxgrey, 2021-09-29:
There may be "tasks" that need to run indefinitely or permanently. For example, a patrolling task or a continuous delivery task (where we expect a new delivery to always be ready to go as soon as the previous delivery is finished).

In these cases, it's best to assign the continuous task to one robot, and then exclude the robot from being assigned any further tasks until the indefinite task is canceled or terminates some other way. (Note that we'll assume these indefinite tasks will be responsible for having battery recharging phases)

For the planner, we could indicate that a task is indefinite by changing [`Estimate::finish_state`](https://github.com/open-rmf/rmf_task/blob/e1c2d51f7f32daf36e5298247656ed1e84b667bd/rmf_task/include/rmf_task/Estimate.hpp#L52) to return a `std::optional<State>` where a `std::nullopt` value means that it will run indefinitely (as opposed to [`Task::Model::estimate_finish`](https://github.com/open-rmf/rmf_task/blob/e1c2d51f7f32daf36e5298247656ed1e84b667bd/rmf_task/include/rmf_task/Task.hpp#L142) where returning `std::nullopt` means that the task cannot be performed).

arjo129, 2021-09-30:
Hmm... What about charging? Or do we just ignore that for now.

mxgrey, 2021-09-30:
Since it's difficult to spot and a very important point, I'll highlight it here:

> Note that we'll assume these indefinite tasks will be responsible for having battery recharging phases